socialwifi/RouterOS-api #24: 'utf8' codec can't decode byte 0xbc in position 0: invalid start byte

Kinganmi, 2017-06-13:
```
    def get_ros_info(self):
        api = ros_api.get_api()
        list_queues = api.get_resource(self.dhcp_profix)
        logger.info(list_queues)
        try:
            user_info = list_queues.get()
            logger.info(user_info)
        except Exception as e:
            logger.info(e)
```

2017-06-13 09:21:30,746 [Thread-6:140061903419136] [nam:81] [toolkit:get_ros_info] [INFO]- RouterOsResource(/ip/dhcp-server/lease/)
2017-06-13 09:21:31,102 [Thread-6:140061903419136] [nam:87] [toolkit:get_ros_info] [INFO]- 'utf8' codec can't decode byte 0xbc in position 0: invalid start byte

Pls help me ...  Tks..

kramarz, 2017-06-13:
If you don't want everything to be converted to unicode, then you can write
```
import collections
import routeros_api.api_structure
structure = collections.defaultdict(routeros_api.api_structure.BytesField)
list_queues = api.get_resource(self.dhcp_profix, structure=structure)
```
probably you can still automatically decode all fields except HOST-NAME. Than  you can set structure like:
```
import collections
import routeros_api.api_structure
structure = collections.defaultdict(routeros_api.api_structure.StringField)
structure['host-name']=routeros_api.api_structure.BytesField()
list_queues = api.get_resource(self.dhcp_profix, structure=structure)
```

Kinganmi, 2017-06-14:
```
    def transform_item(self, item):
        key, value = item
        value = value.decode('gbk')
        return (key, self.structure[key].get_python_value(value))
```
Because there are ascii.....

My handling method, 

add   `value = value.decode('gbk')`

**If there is a better solution, please reply**

**Thank you very much for your reply, TKS**

closed 2017-07-24 by Kinganmi