socialwifi/RouterOS-api #27: Send commands with args

andreytsushima, 2017-12-22:
Hi, how can i send a command with args like 
api.get_resource('/tool/wol/interface=bridge-lan, mac='+str(mac))
Thanks

oortega, 2018-01-27:
Hello,

I do not know what is "tool/wol", but follow this:

wol = api.get_resource("/tool/wol")

wol_intreface = wol.get(interface="bridge-lan", mac=str(mac))

This the wiki https://github.com/socialwifi/RouterOS-api/wiki/How-to-use

Regards.

gerarivero, 2018-02-15:
Hello, first of all the api is great second excusme for my english is not my native language. I have the same problem, wol is a function that take two parameters to wake up a pc on your lan.
The same thing happends with the monitor-traffic inside interface, i can't monitor an interface in real time.
When i run 
monitor= api.get_resource("/interface/monitor-traffic")
monitor.get(interface='ether1')
This raises the next **error (Error "no such command" executing command b\ '/interface/monitor-traffic/print ?interface=LAN .tag=3\ ''no such command')**

Thanks in advance

stevehaskew, 2018-06-21:
I've added a section on the wiki with an example of how I managed to do this.

gera-rivero, 2018-06-21:
Excellent just pass your key values as binary and works perfect. Greets from Argentina

jgoclawski, 2018-08-23:
Closing this as we have this documented now.
https://github.com/socialwifi/RouterOS-api/wiki/How-to-use#execute-commands
Thanks @stevehaskew and @oortega!

closed 2018-08-23 by jgoclawski