socialwifi/RouterOS-api #30: "SSL" support

davidc, 2018-06-20:
With the upcoming removal of hashing the API login credentials in 6.43, it will be more important than ever to have support to access the API over SSL.

Ideally with these two optional features as well:
- verify the certificate hostname matches what we are connecting to
- verify the server's certificate is signed by a user-supplied CA certificate

stevehaskew, 2018-06-22:
I have taken some existing SSL work and finished it up with the above options. I was also looking for this. Submitted PR #31

stevehaskew, 2018-07-24:
This is now fixed and SSL support now in master.

jgoclawski, 2018-08-23:
Fixed in #31.
Released as 0.15.0.
Thanks a lot to everyone involved! @stevehaskew @kramarz @davidc

closed 2018-08-23 by jgoclawski