socialwifi/RouterOS-api #35: Can api-ssl be used without a cert?

terminator14, 2018-08-07:
According to this short page: https://wiki.mikrotik.com/wiki/Manual:API-SSL

it seems that it is possible to talk to a Mikrotik using api-ssl WITHOUT a certificate being configured under IP --> Services --> api-ssl --> certificate.

Am I understanding that right? And is your routeros-api software able to do this? Or is it necessary to first login to the Mikrotik, create a self-signed certificate, and set api-ssl to use it?

When I tried this without having a certificate set for api-ssl to use, I got this error:

`ssl.SSLError: [SSL: SSLV3_ALERT_HANDSHAKE_FAILURE] sslv3 alert handshake failure (_ssl.c:833)`

Thanks

jgoclawski, 2018-08-23:
I haven't tried it, but you can configure the SSL Context and pass it to API. You can set any options you need, including anonymous DH as cipher which is disabled by default I think.
Docs: https://docs.python.org/3.7/library/ssl.html#ssl-contexts

SSL configuration is documented in wiki - https://github.com/socialwifi/RouterOS-api/wiki/How-to-use#using-ssl and it was added in #31.
This is a new feature, just released on pypi as version 0.15.0.

closed 2018-08-23 by jgoclawski

MajesticFalcon, 2020-03-13:
I hate to comment on such an old issue, but this seems to be a thorn in my side. Reading through the SSL context documentation and the Social Wifi Routeros documentation, I still can't get anon Diffie Helman to work correcly. Is there any way you could test it on your side to see if it even works?

eduardomazolini, 2020-08-24:
in api_socket.py
try use code below, work for me.

    if ssl_context is not None:
        ssl_context.set_ciphers("ADH:ALL:@SECLEVEL=0")
        api_socket = ssl_context.wrap_socket(api_socket,server_hostname=hostname)
    return SocketWrapper(api_socket)

eduardomazolini, 2020-08-24:
Or create your owner context
            import ssl
.
.
.
            ssl_context = ssl.create_default_context()
            ssl_context.check_hostname = False
            ssl_context.verify_mode = ssl.CERT_NONE
            ssl_context.set_ciphers("ADH:ALL:@SECLEVEL=0")
            connection = routeros_api.RouterOsApiPool(ip_address,
                                                      username=cred["username"],
                                                      password=cred["password"],
                                                      plaintext_login=plaintext_login,
                                                      ssl_context=ssl_context)