socialwifi/RouterOS-api #40: Parameter after print

marcelosantoro, 2019-02-13:
Hi friends.... 

I Would like to know... how can i send parameters to "list = api.get_resource()" or "list.get()" commands

Example
/interface print stats-detail
or
/ppp active print count-only

How can i get above results with RouterOS-api ???

marcelosantoro, 2019-04-13:
nobody ??? :(

fernandolcx, 2019-06-10:
Have you figured out how to do it?

marcelosantoro, 2019-06-10:
> Have you figured out how to do it?

Cara infelizmente ainda não consegui. :(

fernandolcx, 2019-06-10:
Está tudo certo:
```python
result = api.get_resource('interface/pppoe-server').call('print', arguments={'count-only': ''}, queries={'service': 'service10'})
print(json.dumps(result, indent=4))
```
mas retorna vazio.
Se alterar `count-only` para `detail` funciona.

Behoston, 2019-09-24:
THX @fernandolcx for the solution.
I'm closing this issue for now.
If you still have questions, feel free to reopen/comment.

closed 2019-09-24 by Behoston