socialwifi/RouterOS-api #42: Explanations about correct using of 'structure' paramter of RouterOsApi.get_resource() method

bobr-kun, 2019-06-10:
To some extent my question is related to this one: https://github.com/socialwifi/RouterOS-api/issues/40
But not so narrow. Basically I want to figure out how to use queries of RouterOS API **print** command(which is slightly differs from regular RouterOS CLI **print** command). I think that this RouterOS API implementation is quite able to do that. 
So, when forming a '**resource**'(speaking in the terms of current RouterOS API implementation) with **RouterOsApi.get_resource()** method to be queried from RouterOS with **RouterOsResource.get()**, sometimes I want to make it as much narrow as possible to get as much less data as possible(ideal option - to get exactly that record I need and no others). So, when forming such a '**resource**' object I want to add some conditional parameters and I think that '**structure**' parameter is designed for this exactly(or am I wrong?).
Finally, I'm asking the developers of this RouterOS API implementation: If you'd be so kind to shed some light on my question and give some more specific explanations to us about the use of that '**structure**' parameter or how is it possible to make queries with this RouterOS API implementation?

fernandolcx, 2019-06-10:
I spent my weekend trying to figure out how to do it exactly what you said.

With [tikapy](https://github.com/vshn/tikapy) it's neat like `pprint(api.talk(['/interface/pppoe-server/print', '=count-only'], '?service=service4'))`, but with this one I can't.

bobr-kun, 2019-06-11:
Thanks for your answer @fernandolcx 
But may be the developers will shed some light? But seems they just disappeared...