socialwifi/RouterOS-api #45: Last RouterOS update broke login process

okazdal, 2019-07-02:
Hello,
It seems like with latest RouterOS updates, login process needs to be modified.
Currently, we are getting this message on our script previously working:

Traceback (most recent call last):
  File "C:/Users/merlyn/PycharmProjects/violence/mikrotik/dhcp_lease.py", line 11, in <module>
    api = connection.get_api()
  File "C:\Users\merlyn\Anaconda3\lib\site-packages\routeros_api\api.py", line 51, in get_api
    self.api.login(self.username, self.password, self.plaintext_login)
  File "C:\Users\merlyn\Anaconda3\lib\site-packages\routeros_api\api.py", line 93, in login
    'login', {'name': login.encode(), 'response': hashed})
  File "C:\Users\merlyn\Anaconda3\lib\site-packages\routeros_api\resource.py", line 39, in call
    additional_queries=additional_queries).get()
  File "C:\Users\merlyn\Anaconda3\lib\site-packages\routeros_api\api_communicator\encoding_decorator.py", line 31, in get
    response = self.inner.get()
  File "C:\Users\merlyn\Anaconda3\lib\site-packages\routeros_api\api_communicator\async_decorator.py", line 17, in get
    self.response = self.receiver.receive(self.tag)
  File "C:\Users\merlyn\Anaconda3\lib\site-packages\routeros_api\api_communicator\exception_decorator.py", line 19, in receive
    self.handle_exception(e)
  File "C:\Users\merlyn\Anaconda3\lib\site-packages\routeros_api\api_communicator\exception_decorator.py", line 37, in handle_exception
    raise exception
  File "C:\Users\merlyn\Anaconda3\lib\site-packages\routeros_api\api_communicator\exception_decorator.py", line 17, in receive
    return self.inner.receive(tag)
  File "C:\Users\merlyn\Anaconda3\lib\site-packages\routeros_api\api_communicator\key_cleaner_decorator.py", line 14, in receive
    answers = self.inner.receive(tag)
  File "C:\Users\merlyn\Anaconda3\lib\site-packages\routeros_api\api_communicator\base.py", line 50, in receive
    raise response.error_as_exception
routeros_api.exceptions.RouterOsApiCommunicationError: ('Error "invalid user name or password (6)" executing command b\'/login =name=admin =response=0058cb2f3757642cab91e77913fe034363 .tag=2\'', b'invalid user name or password (6)')

Process finished with exit code 1

Behoston, 2019-07-02:
Please provide more information - we need exact RouterOS version to investigate.
Use: `/system resource print`

jgoclawski, 2019-07-02:
@okazdal Please check our Wiki regarding newer ROS versions: https://github.com/socialwifi/RouterOS-api/wiki/How-to-use#login-for-routeros-v643-onwards

If you have upgraded your device's ROS version, please update also this library and use the `plaintext_login=True`.

Closing for now.

closed 2019-07-02 by jgoclawski

scramatte, 2019-08-21:
With routeros 6.44.5  it doesn't works ...
I've got following error :

`connection = routeros_api.RouterOsApiPool('172.21.4.1', username='api', password='apiro',plaintext_login=True)
api = connection.get_api()
`

`Traceback (most recent call last):
  File "..\Custom Sensors\python\test.py", line 6, in <module>
    api = connection.get_api()
  File "C:\Program Files (x86)\PRTG Network Monitor\Python34\lib\site-packages\r
outeros_api\api.py", line 51, in get_api
    self.api.login(self.username, self.password, self.plaintext_login)
  File "C:\Program Files (x86)\PRTG Network Monitor\Python34\lib\site-packages\r
outeros_api\api.py", line 82, in login
    response = self.get_binary_resource('/').call('login',{ 'name': login, 'pass
word': password })
  File "C:\Program Files (x86)\PRTG Network Monitor\Python34\lib\site-packages\r
outeros_api\resource.py", line 39, in call
    additional_queries=additional_queries).get()
  File "C:\Program Files (x86)\PRTG Network Monitor\Python34\lib\site-packages\r
outeros_api\resource.py", line 45, in call_async
    additional_queries=additional_queries)
  File "C:\Program Files (x86)\PRTG Network Monitor\Python34\lib\site-packages\r
outeros_api\api_communicator\encoding_decorator.py", line 12, in call
    path, command, arguments, queries, additional_queries)
  File "C:\Program Files (x86)\PRTG Network Monitor\Python34\lib\site-packages\r
outeros_api\api_communicator\async_decorator.py", line 6, in call
    tag = self.inner.send(*args, **kwargs)
  File "C:\Program Files (x86)\PRTG Network Monitor\Python34\lib\site-packages\r
outeros_api\api_communicator\exception_decorator.py", line 11, in send
    return self.inner.send(*args, **kwargs)
  File "C:\Program Files (x86)\PRTG Network Monitor\Python34\lib\site-packages\r
outeros_api\api_communicator\key_cleaner_decorator.py", line 11, in send
    queries=encoded_queries, additional_queries=additional_queries)
  File "C:\Program Files (x86)\PRTG Network Monitor\Python34\lib\site-packages\r
outeros_api\api_communicator\base.py", line 19, in send
    self.send_command(command)
  File "C:\Program Files (x86)\PRTG Network Monitor\Python34\lib\site-packages\r
outeros_api\api_communicator\base.py", line 37, in send_command
    self.base.send_sentence(command.get_api_format())
  File "C:\Program Files (x86)\PRTG Network Monitor\Python34\lib\site-packages\r
outeros_api\sentence.py", line 59, in get_api_format
    formated.append(b'=' + key + b'=' + value)
TypeError: can't concat bytes to str`

okazdal, 2019-08-22:
You should check the other issue with suggested fix.
https://github.com/socialwifi/RouterOS-api/issues/47#issuecomment-519829454