socialwifi/RouterOS-api #49: There is a way to remove elements massively?

rogerarjona, 2019-08-03:
There is a way to remove elements massively?

Hello.
I would like to know if there is a way to delete elements sending a list of ids.

Normally we delete many elements like this:

```
for id_queue in id_list:
	list_queues.remove(id=id_queue)
```
But, there is, or is possible something like this?

```
id_list = ['*1', '*2', '*3', '*4']
list_queues.remove(id_list)
```

stevehaskew, 2019-08-14:
As far as I am aware there is not a way to do that currently. All my contributions so far have been around the initial connection etc. and I haven't delved into the resource objects and how to work with/modify them. Unfortunately I don't have the time at the moment to do so.

I like the idea - it makes sense to be able to provide a list and the remove function detect it and act accordingly. Feel free to make a contribution yourself if you are able!

jiri-lunacek, 2021-02-27:
> Hello.
> I would like to know if there is a way to delete elements sending a list of ids.
> 
> Normally we delete many elements like this:
> 
> ```
> id_list = ['*1', '*2', '*3', '*4']
> for id_queue in id_list:
> 	list_queues.remove(id=id_queue)
> ```
> 
> But, there is, or is possible something like this?
> 
> ```
> id_list = ['*1', '*2', '*3', '*4']
> list_queues.remove(id_list)
> ```
> 
> @jgoclawski , @stevehaskew

Hello.
I just came to this project. And found that the following approach works:

```
id_list = ['*1', '*2', '*3', '*4']
list_queues.remove(','.join(id_list))
```

stevehaskew, 2021-02-27:
Thanks for taking the time to share this solution, it's much appreciated!