socialwifi/RouterOS-api #53: api_structure.py utf-8 code

Jerning, 2019-10-15:
UnicodeDecodeError                        Traceback (most recent call last)
<ipython-input-6-b015b4e25dcf> in <module>
----> 1 for i in a:
      2     if i['server'] == 'dhcp_lan':
      3         print(i)
      4

~/ros_api/core/routeros_api/resource.pyc in transform_dictionary(self, row)

~/ros_api/core/routeros_api/resource.pyc in <genexpr>(.0)

~/ros_api/core/routeros_api/resource.pyc in transform_item(self, item)

~/ros_api/core/routeros_api/api_structure.pyc in get_python_value(self, bytes)

UnicodeDecodeError: 'utf-8' codec can't decode byte 0xba in position 0: invalid start byte

One of the records contains Chinese

Behoston, 2019-10-15:
Probably you are using ISO-8859-1 instead of UTF-8.
We are open to pull requests :) 
You can also write test case that reproduce bug.

Matroskin-omsk, 2020-01-03:
hello use comment "тестовый коммент", and get command  return this comment
probleme in routeros_api/api_structure.py
    def get_python_value(self, bytes):
        return bytes.decode() <- you must be able to specify an encoding other than utf-8

Jerning, 2020-01-04:
yes，the probrom in api_structure.py

I temporarily modified:

class StringField(object):
    def get_mikrotik_value(self, string):
        return string.encode()

    def get_python_value(self, bytes):
        try:
            return bytes.decode()
        except Exception as e:
            return 'value has chinese'

Jerning Chan

voytec69, 2020-04-28:
I confirm, my fix:
`31c31
<             return bytes.decode()
---
>             return bytes.decode("utf-8", "ignore")`

Navisho, 2020-10-30:
Hola!
like @voytec69 changing  api_structure.py but doing the same thing in encode rutine.  This fix the spanish chracters when change the comments.

```
class StringField(Field):
    def get_mikrotik_value(self, string):
        return string.encode("cp1252", "ignore")

    def get_python_value(self, bytes):
        return bytes.decode("cp1252", "ignore")

```

rahmadsandy, 2021-11-10:
close issue with
https://github.com/socialwifi/RouterOS-api/issues/68

closed 2025-03-07 by jgoclawski