socialwifi/RouterOS-api #54: How to send .proplist param?

evpamex, 2019-10-26:
Please explain how can I send filter .proplist with API requests?

This is not working:
list_address =  api.get_resource('/interface')
pprint(list_address.get(.proplist="name"))

originar API requests would look like this:
<<< /interface/print
<<< =.proplist=name

evpamex, 2019-10-26:
and this also not working:
list_address = api.get_resource('/interface')
pprint(list_address.get(proplist="name")) # without dot

I just get empty output.

Matroskin-omsk, 2020-01-04:
hello 
there is a solution?

mlow, 2020-04-06:
There is a solution.

```python
queues = api.get_resource('/queue/simple')
queues.call('print', { '.proplist': '.id,name,target' }, { 'max-limit': '1000000/100000' })
```