socialwifi/RouterOS-api #57: Get wifi signal correct syntax

Summer88, 2020-03-03:
Hi,
thanks for your work. this library is really wonderful!
I need to get signal strenght from a wireless client, if I save the client in access list I can get it with it's comment, but it must be dynamic, so how can I get the signal by requesting last-ip?

```
import routeros_api
connection = routeros_api.RouterOsApiPool('192.168.1.1', plaintext_login=True)
api = connection.get_api()
list = api.get_resource('/interface/wireless/registration-table')

# this works fine:
list.get(comment="THISWORKS")[0]['signal-strength']
```

This doesn't work:
```
list.get(last-ip="192.168.1.11")
```
with this error:
```
            ^
SyntaxError: keyword can't be an expression

```

Thanks, BR

igoras1993, 2020-05-12:
This is actually a SyntaxError, You are writing a bad python code. Here:
```
list.get(last-ip="192.168.1.11")
```
a `last-ip` looks like someone wants to substract (a `-` sign) two variables. This syntax is incorrect and this is why You cannot name Your variables with a `-` sign, also inside a keyword declaration.

You actually missed the note in readme that stays:

> NOTE: Atributes with -, like max-limit use underscore _: max_limit

So You should write it like that:
```
list.get(last-ip="192.168.1.11")
```
But You have been waiting for response over 2 months, @socialwifi is this project still maintained?

jgoclawski, 2020-05-12:
@igoras1993 thanks for helping out. The project is maintained, but unfortunately we don't have resources to provide support and answer all questions. We've shared the code that we're using internally hoping that someone else will find it useful. See https://github.com/socialwifi/RouterOS-api/issues/19#issuecomment-111830885

igoras1993, 2020-05-12:
@jgoclawski thanks for response, acknowledged :)

closed 2021-11-03 by jgoclawski