socialwifi/RouterOS-api #60: how to get monitor-traffic using this?

adhiartawika, 2020-06-16:
i try doing this content = api.get_content("/interface/monitor-traffic interface=ether1") but it wont work can anyone help me? or is there any other way that i can get the tx or rx bits per second? thanks

Nixellion, 2020-06-24:
Came here to ask the same thing!
I also tried:

```
api.get_binary_resource('/interface').call('monitor-traffic', {'name': b'lte'})
api.get_binary_resource('/interface').call('monitor-traffic lte1,aggregate')
api.get_binary_resource('/').call('interface monitor-traffic lte1,aggregate')
```

And I get:

```
  File "C:\Python37\lib\site-packages\routeros_api\resource.py", line 39, in call
    additional_queries=additional_queries).get()
  File "C:\Python37\lib\site-packages\routeros_api\api_communicator\encoding_decorator.py", line 31, in get
    response = self.inner.get()
  File "C:\Python37\lib\site-packages\routeros_api\api_communicator\async_decorator.py", line 17, in get
    self.response = self.receiver.receive(self.tag)
  File "C:\Python37\lib\site-packages\routeros_api\api_communicator\exception_decorator.py", line 19, in receive
    self.handle_exception(e)
  File "C:\Python37\lib\site-packages\routeros_api\api_communicator\exception_decorator.py", line 37, in handle_exception
    raise exception
  File "C:\Python37\lib\site-packages\routeros_api\api_communicator\exception_decorator.py", line 17, in receive
    return self.inner.receive(tag)
  File "C:\Python37\lib\site-packages\routeros_api\api_communicator\key_cleaner_decorator.py", line 14, in receive
    answers = self.inner.receive(tag)
  File "C:\Python37\lib\site-packages\routeros_api\api_communicator\base.py", line 50, in receive
    raise response.error_as_exception
routeros_api.exceptions.RouterOsApiCommunicationError: ('Error "no such command" executing command b\'/interface/monitor-traffic lte1,aggregate .tag=3\'', b'no such command')
```

Nixellion, 2020-06-24:
I think this sort of works:

https://github.com/socialwifi/RouterOS-api/issues/27

```
print (api.get_binary_resource('/').call('interface/monitor-traffic', {'interface': b'lte1'}))
```

Only it hangs as it seems it goes into monitoring loop, without returning result. I wonder how to make it just return current tx rx values

adhiartawika, 2020-06-25:
i think this will work dude:

monitor_traffic = api.get_binary_resource('/interface').call('monitor-traffic',
                                                             {'interface': b'ether1', 'once': b' '})
ftx = monitor_traffic[0]['tx-bits-per-second']
frx = monitor_traffic[0]['rx-bits-per-second']
d = ftx.decode()
a = frx.decode()
ctx_interface = {'ftx': d, "frx": a}
print(ctx_interface)
i do this so it reurn de ftx and frx

Nixellion, 2020-06-25:
@adhiartawika Awesome, thanks, never would've guessed it's just empty param 🗡️ 

I think docs could use a bit of improvement on how to operate such things.

closed 2021-11-03 by jgoclawski