socialwifi/RouterOS-api #68: Unicode Decode Error (UTF-8)

l-portugal, 2020-12-02:
My script obtain a list of DHCP Server leases. When i run the script, this show the error: 

_**"UnicodeDecodeError: 'utf-8' codec can't decode byte 0xf1 in position 6: invalid continuation byte"**_

the command is : 

cmd = api.get_resource('/ip/dhcp-server/lease')
leases = cmd.get()

In leases, some characters in "host-name" field they may be: á,é,í,ó,ú 

is there any way to fix this error without change the characters in host-name field?

The script is in python3.

Thanks.

skeapskeap, 2020-12-03:
Try to change decode params of method get_python_value in routeros_api/api_structure.py
There is _bytes.decode()_ by default. Try _bytes.decode('cp1251')_
```
def get_python_value(self, bytes):
        return bytes.decode('cp1251')
```

l-portugal, 2020-12-03:
> Try to change decode params of method get_python_value in routeros_api/api_structure.py
> There is _bytes.decode()_ by default. Try _bytes.decode('cp1251')_
> 
> ```
> def get_python_value(self, bytes):
>         return bytes.decode('cp1251')
> ```

Thanks  skeapskeap. I solve it changing **_bytes.decode()_** for **_bytes.decode("utf-8", "ignore")_**

routeros_api/api_structure.py:

```
def get_python_value(self, bytes):
        return bytes.decode("utf-8", "ignore")
```

thanks

colinhd8, 2021-04-30:
I have the same issue, thank you for your comment.

> > Try to change decode params of method get_python_value in routeros_api/api_structure.py
> > There is _bytes.decode()_ by default. Try _bytes.decode('cp1251')_
> > ```
> > def get_python_value(self, bytes):
> >         return bytes.decode('cp1251')
> > ```
> 
> Thanks skeapskeap. I solve it changing **_bytes.decode()_** to **_bytes.decode("utf-8", "ignore")_**
> 
> routeros_api/api_structure.py:
> 
> ```
> def get_python_value(self, bytes):
>         return bytes.decode("utf-8", "ignore")
> ```
> 
> thanks

Gunni, 2023-10-02:
I had the same problem, patched it on my side using: `return bytes.decode('iso-8859-1')` but would like an official fix.

montvid, 2024-10-30:
yep, a bug - worked for me:
```
class StringField(Field):
    def get_mikrotik_value(self, string):
        return string.encode()

    def get_python_value(self, bytes):
        return bytes.decode("utf-8", "ignore")
```

closed 2025-03-07 by jgoclawski