socialwifi/RouterOS-api #77: How sign certificate via API?

soksssnn, 2021-11-02:
Hello! 
Please help me. I can't find how I can sign certificate via API. Cerificate was create. I tried do it in that way:
api = connection.get_api()
api.get_resource('/certificate').sign(name=user_ovpn,ca='ovpn_ca')
But got AttributeError: 'RouterOsResource' object has no attribute 'sign'.

Sorry for the stupid question. I am a newcomer in the python.
Thank you.

jgoclawski, 2021-11-03:
Did you try the examples from the docs? https://github.com/socialwifi/RouterOS-api#execute-commands

igorvukotic, 2022-02-24:
Hi,
I'm strugling with `add-scep template=client_tpl scep-url="http://xxx.xxx.xxx.xxx/scep/ca"`
i tryed with c.add-scep(..., c.add_scep(...
but no result, please help