socialwifi/RouterOS-api #79: hoe can I get 'count-only' for files?

SmartTransp, 2021-12-23:
I'm trying to get jsut the amount of files i ahve in the router 
on the CLI I'm running this command:
`
:put [file print count-only ]
`

how can I do the same in the api?
I have try this 
`
replay = api.get_resource('/file/')
test_answer = replay.get(print='count-only')
print('how many files? - ', test_answer 
`
but I get an empty response
`how many files?  -  []`

SmartTransp, 2021-12-26:
I have also try this 
`
test = api.get_binary_resource('/').call('file', {'print': 'count-only'})
`

but I get error 
can't concat str to bytes

so I check the test command in the Readme file 
`api.get_binary_resource('/').call('ping', { 'address': '192.168.56.1', 'count': '4' })`

get the same error 

so what is missing?

Thanks,

M0r13n, 2022-08-19:
I also struggled with this issue. I wanted to get the number of connections - but not details about all connections. I messed around with this module and took a look at the source code. The important piece for me to understand was, that the API returns the count as the return code when using `print-only`. This is a custom attribute of the response class, which looks like a list and behaves like a list, but it not a list :-)

```py
import routeros_api

connection = routeros_api.RouterOsApiPool()
api = connection.get_api()

answer = api.get_binary_resource('/ip/firewall/connection/').call('print', {'count-only': b''})
# answer looks and feels like an empty list: [], but is has a special attribute `done_message`
done_message = answer.done_message

# `done_msg` is a dict with the return code as a key - which is the count that we are looking for
cnt = done_message['ret'].decode()
print(cnt)
```

You can adapt the example to your case.

akpw, 2024-05-02:
this one does not work as expected though:
```
st = api.get_resource('/ip/route/').call('print', {'count-only':'', 'where': 'static'})            
print(st.done_message)
```
nor this:
```
st = api.get_resource('/ip/route/').call('print', {'count-only':''}, {'where': 'static'})            
print(st.done_message)
```

@jgoclawski by chance any advice on how to use this approach with the 'where' clause?

jgoclawski, 2025-03-07:
Thanks for raising this. The solution should be something like this:
```
api.get_resource('/ip/route').call('print', {'count-only': ''}, {'static': 'yes'}).done_message
```

I've added an example to the README.

closed 2025-03-07 by jgoclawski