socialwifi/RouterOS-api #83: Monitor ethernet and get negotiated rate

cykyy, 2022-04-18:
Hello, I am trying to get the negotiated rate of the interface. This command ("/interface ethernet monitor ether1 once") gives the exact information I need on Mikrotik CLI but it doesn't run through the API! Can anyone help me with how I can get the information?.
I really appreciate any help you can provide 🙏.

cykyy, 2022-04-18:
I have just fixed this,
this did the trick, `conn.get_binary_resource('/').call('interface/ethernet/monitor', {'numbers': f'{eth.get("name")}'.encode("utf-8"), 'once': 'once'.encode("utf-8")})`

closed 2022-04-18 by cykyy

banderlog, 2023-10-10:
@cykyy thanks, you saved me) But from where you get this necessary parameters in dict, is there some doc about it?