socialwifi/RouterOS-api #89: New Release / Tag

jvanderaa, 2023-05-26:
Looks like this library has not had a release/tag in a while. Yet there was a merge in March. The last "documented" release supported is Python 3.7 that is end of life shortly. It is probably worth updating the library/putting out a release.

irrwitzer42, 2024-04-20:
Yes, please "release" a new tag if possible - I'm using mktxp which is depending on the pip module, which is stuck on the latest release 0.17.
Since most of my routers are IPv6 only, I'm very much looking forward to the latest fix from march 2023...
Thanks in advance!

And BTW: thanks so much for this project, which makes life so much easier! Keep on the great work!

jgoclawski, 2024-04-30:
@jvanderaa @irrwitzer42 Thanks for bringing this up. We've just released version 0.18 which includes the IPv6 feature and  updated Python versions support.

closed 2024-04-30 by jgoclawski