socialwifi/RouterOS-api #92: "missing =.id=" on `poe set` on RouterOS 7.11.2

banderlog, 2023-10-24:
Hi

When I am running this on RouterOS 7.11.2:

```python
connection = routeros_api.RouterOsApiPool(IP, username=USER,  password=PWD, plaintext_login=True)
api = connection.get_api()
api.get_resource('/interface/ethernet/poe').call('set', {'name': port_name, 'poe-out': 'off'})
```

I get next error:

```python
RouterOsApiCommunicationError: ('Error "missing =.id=" executing command b\'/interface/ethernet/poe/set =name=ether5 =poe-out=off .tag=3\'', b'missing =.id=')
```

With RouterOS 7.9.2 everything is fine. Please help

banderlog, 2024-07-08:
For RouterOs  6.49.15 (2024-Apr-24 16:04) one must use `.id` only, or command will just not work without any error

`api.get_resource('/interface/ethernet/poe').call('set', {'poe-out': 'forced-on', ".id": "*2"})`

jgoclawski, 2025-03-07:
Thanks, I'll put something similar in the wiki for others looking to do this. You can use either '.id' or 'numbers' like in CLI:
```python
api.get_resource('/interface/ethernet/poe').call('set', {'poe-out': 'off', 'numbers': '0'})
```